MCUdude/MightyCore #225: Can't burn bootloader to 324PB

scrouthtv, 2021-10-01:
Hello,

I am currently trying to burn the bootloader to an atmega 324PB using an Arduino Uno.

It is wired up like this:
[image]
On my PCB, each of the capacitors is directly behind one of the Vcc pins of the IC.

I am not able to burn the bootloader using an Arduino Uno.
I connected
 - GND of the PCB to GND of the Uno
 - Vcc of the PCB to 5V of the Uno
 - SCK of J2 to SCK of the ICSP header of the Uno
 - MOSI of J2 to MOSI of the ICSP header
 - MISO of J2 to MISO of the ICSP header
 - Reset of J1 to Pin 10 of the Uno (with no capacitor in between nor a pullup)

If I try to upload, I get this error
<details>
C:\Users\lenni\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\lenni\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.3/avrdude.conf -v -patmega324pb -cstk500v1 -PCOM12 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0b11110101:m -Uhfuse:w:0b11010110:m -Ulfuse:w:0b11111111:m 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\lenni\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.3/avrdude.conf"

         Using Port                    : COM12
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega324PB
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       1024    4      0  9000  9000 0xff 0xff
           flash         33     6   256    0 yes     32768  128    256  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000 (retrying)

Error while burning bootloader.
Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.
</details>

I triple-checked all soldered connections and parts. Did I wire up something wrong?

Is it possible that the chip got destroyed during reflow? I used a hot-air gun set to 300C and fixed some things manually with 350C.

Do you have any ideas how I would go about troubleshooting?

tomaskovacik, 2021-10-01:
miso to mosi

scrouthtv, 2021-10-01:
> 
> 
> in old times, there was required 100uF capacitor on RESET line of arduino which runs ArduinoISP, try it

I thought it was a 100 nF + a 10k Pullup on the slave side

tomaskovacik, 2021-10-01:
100u to ground to disable autoreset

scrouthtv, 2021-10-01:
With the capacitor, I tried to burn it 4 times, and one out of the four worked. The other three runs gave me the same error.
Any idea why it's so inconsistent?

tomaskovacik, 2021-10-01:
strange, you can try disable auto-reset on UNO board the correct way: cutting the trace in pad marked "RST EN"
[image]

scrouthtv, 2021-10-01:
I also tried burning with a bare atmega 328 board with SS of the master connected to reset of the slave

scrouthtv, 2021-10-03:
> cutting the trace in pad marked "RST EN"

Same error

tomaskovacik, 2021-10-06:
they can also try my old trick, select internal oscillator 1 or 8MHz and burn the bootloader, then select external 16MHZ crystal and burn new bootloader again but I do not think it will help.

scrouthtv, 2021-10-07:
> is there a 16 MHz sine wave on the XTAL1 pin?

Yes. minimum 620, maximum 840 mV. 

> How many pF's are the capacitors attached to the crystal?

I'm using [this](https://www.mouser.de/ProductDetail/81-CSTNE16M0V530000R) resonator. The datasheet doesn't specify a capacitance, but according to [this product listing](https://www.murata.com/~/media/webrenewal/support/library/catalog/products/timingdevice/ceralock/p17e.ashx), it has *16 pF*

> Is the ATmega324PB brand new

Yes. I guess I destroyed it during reflow?

tomaskovacik, 2021-10-08:
oh, you have usasp, then select usbasp slow, and try that, and also you can try to select  "internal oscillator 1Mhz". To be honest, I did not study what it changes, but it works for me to boot up new chips, so I do not care  :)))

Hamza-arch, 2021-10-10:
I have been able to upload the bootloader using a new atmega16A chip. I think I have made some mistakes on last one. So, I might need to reflash it. I was searching it and I think it is easy to do using avrdudess but I was not able to find out the fuse settings to flash. If someone has a good tutorial regarding it, it can be very helpful.
Also, I am trying to upload code uisng arduino Rx and Tx pins. I have not been successful in this regard. It is giving the errors of not in sync. I have removed the atmega chip from arduino uno. However, arduino is still reading it as arduino uno. Can anyone explain it.

rocifier, 2022-09-28:
What are your crystal oscillator cap values? The datasheet gives a very specific range for the fuse selector you've chosen:

[image]

rocifier, 2022-09-28:
Sorry just confirmed your resonator part is 16pF so that's fine

rocifier, 2022-09-28:
I think probably the reason is that you need to power it externally and not depend on the power that comes from the Arduino Uno as ISP

scrouthtv, 2022-09-28:
> the power that comes from the Arduino Uno as ISP

Shouldn't that simply be the 5V from USB? And that power should be fine as it is always on as soon as you plug it into the PC, right?

rocifier, 2022-09-28:
> 

from my experience it's not always enough to power the programmer and the board at the same time during programming from USB

MCUdude, 2023-07-25:
I'm closing this issue because it's most likely a hardware issue. I've designed a commercial product in the past, based on an aTmega324PB. I've been having _so much trouble_ with the external oscillator because the 324PB isn't strong enough to properly drive a crystal that's slightly too capacitive. After several board revisions, I ended up replacing the crystal and load capacitors with a dedicated clock that outputs a clean clock signal that is wired straight to the XTAL1 pin on the 324PB. No more issues!

closed 2023-07-25 by MCUdude