MCUdude/MightyCore #226: I2c freezing ATmega1284

leetut, 2021-10-05:
My sketch works using a 328p standard Arduino core, but freezes during startup using a 1284 with mightycore, 
Anyone know how to fix it?
[](https://forum.arduino.cc/t/atmega1284-twi-i2c/911052/19)

MCUdude, 2021-10-05:
The Wire library for MightyCore doesn't have a built-in timeout feature like the one for the standard Arduino core has. However, the reason why it freezes is probably because the i2c slave device is constantly pulling either the SDA or the SCL line low, which is not supposed to happened on an i2c bus.

The reason why I haven't ported over the latest Wire library is that it takes up significantly more RAM and flash.

There is nothing wrong with the Wire library itself. I've used this exact library on a 1284P in multiple commercial devices that's been running for years and years without any issues.

leetut, 2021-10-05:
My Digipot SDA & SCL are pulled high with resistors, 
Also the sketch doesn't use wire, it uses a custom i2c library HW_I@CMaster.h
Arduino forum guys suggested I convert the sketch to use wire, 
but that didn't seem to work or I did it incorrectly,
either way, it works using 328p but not 1284

JensGrabner, 2021-11-27:
In my Project https://github.com/JensGrabner/snc98_Slash-Number-Calculator the I2C works fine on 1284.

MCUdude, 2023-05-19:
Wire timeout functionality is now present, which will prevent this, but it is disabled by default.

Modify this file to enable it:
https://github.com/MCUdude/MightyCore/blob/master/avr/libraries/Wire/src/Wire_timeout.h

If you're using PlatformIO:
https://github.com/MCUdude/MightyCore/blob/master/PlatformIO.md#build_flags

closed 2023-05-19 by MCUdude