MCUdude/MightyCore #231: ER: Add setWireTimeout to the Wire library

mburrack, 2021-12-03:
The main Arduino Wire library has a setWireTimeout method, which is very useful for dealing with unreliable wire connections (which I seem to have on my setup for...reasons?). Regardless, planning for a non-perfect connection in a long-running app is probably wise, which makes setWireTimeout pretty useful.

I've already ported setWireTimeout to the Wire library, but I tried to create a PR for the change and apparently don't have access. I'll see if I can add it to this issue once I submit it...

mburrack, 2021-12-03:
Oh, I attach the file to the message, duh, sorry still new to GitHub.
[src.zip](https://github.com/MCUdude/MightyCore/files/7647184/src.zip)
..

abrahmx, 2022-02-26:
Hello! Can you tell me how can I install this updated library that you are providing? 
Thanks in advance!

MCUdude, 2023-05-19:
Wire timeout functionality is now present, but disabled by default.

Modify this file to enable it:
https://github.com/MCUdude/MightyCore/blob/master/avr/libraries/Wire/src/Wire_timeout.h

If you're using PlatformIO:
https://github.com/MCUdude/MightyCore/blob/master/PlatformIO.md#build_flags

closed 2023-05-19 by MCUdude