MCUdude/MightyCore #235: issue with FastLed?

JoaquimBoavida, 2021-12-29:
Hi
I'm trying to use a Mrga324PB with FastLed library to control a WS2811 chip.
I'm having  an erro "Invalid pin specified" even if I use the default pin 5. Tried other pins, same error.  
Apparently the library checks if pin can be used, and expects some return..
Is there a workaround?

Thanks and sorry for the trouble.

Joaquim

``
In file included from C:\Users\Joaquim\Documents\Arduino\libraries\FastLED\src/FastLED.h:65:0,
                 from C:\Users\Joaquim\AppData\Local\Temp\arduino_modified_sketch_678791\ColorPalette.ino:1:
C:\Users\Joaquim\Documents\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Joaquim\Documents\Arduino\libraries\FastLED\src/FastLED.h:48:0,
                 from C:\Users\Joaquim\AppData\Local\Temp\arduino_modified_sketch_678791\ColorPalette.ino:1:
C:\Users\Joaquim\Documents\Arduino\libraries\FastLED\src/fastpin.h: In instantiation of 'class FastPin<5>':
C:\Users\Joaquim\Documents\Arduino\libraries\FastLED\src/platforms/avr/clockless_trinket.h:100:49:   required from 'class ClocklessController<5, 5, 5, 9, (EOrder)66, 0, false, 10>'
C:\Users\Joaquim\Documents\Arduino\libraries\FastLED\src/chipsets.h:570:7:   required from 'class WS2811Controller800Khz<5, (EOrder)66>'
C:\Users\Joaquim\Documents\Arduino\libraries\FastLED\src/FastLED.h:111:52:   required from 'class WS2811<5, (EOrder)66>'
C:\Users\Joaquim\Documents\Arduino\libraries\FastLED\src/FastLED.h:302:39:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2811; unsigned char DATA_PIN = 5; EOrder RGB_ORDER = (EOrder)66]'
C:\Users\Joaquim\AppData\Local\Temp\arduino_modified_sketch_678791\ColorPalette.ino:40:67:   required from here
C:\Users\Joaquim\Documents\Arduino\libraries\FastLED\src/fastpin.h:210:2: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
``

MCUdude, 2021-12-29:
It looks like FastLED doesn't support ATmega324PB out of the box.

What happens if you replace the following line with this?
https://github.com/FastLED/FastLED/blob/b5874b588ade1d2639925e4e9719fa7d3c9d9e94/src/platforms/avr/fastpin_avr.h#L274

```c++
#elif defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega1284P__) \
|| defined(__AVR_ATmega644P__)     || defined(__AVR_ATmega644A__)  \
|| defined(__AVR_ATmega324PB__)    || defined(__AVR_ATmega324PA__) \
|| defined(__AVR_ATmega324P__)     || defined(__AVR_ATmega324A__)  \
|| defined(__AVR_ATmega164P__)     || defined(__AVR_ATmega164A__)  \
|| defined(__AVR_ATmega32__)       || defined(__AVR_ATmega16__)    \
|| defined(__AVR_ATmega8535__)

```

JoaquimBoavida, 2021-12-29:
Hi,
It does compile now! Thanks. I do have the serial signal on the pin I choose (nice and clean with the expected amplitude). But Nothing happens on the LED's yet, but I'm sure i'ts another problem now. I have to figure that out.

Thanks!

link0687, 2022-05-13:
Hi, how do you modify the file as I will always get an error saying that I don't have permission?

closed 2022-07-24 by MCUdude