MCUdude/MightyCore #237: PWM Output Pin Modification

mclabsud, 2022-01-27:
Dear MCUdude,
I like very much your project. I simply want to ask if there's the possibility to change the PWM pins in your project. I will use the 48 pin version of the chip.
Since the PWM are essentially digital pins, I hope it's possible.

Thank you very much in advise to anyone can help me.

MCUdude, 2022-01-27:
Hi!

Sorry, be the PWM pins cannot be re-routed to different pins. It's a limitation the chip family has. The internal timers that generate the PWM signal are physically connected to the pins.

mclabsud, 2022-01-28:
Ok,

thank you very much for your answer... I was hoping it wasn't like that but ... ok! - I will have to change the PCB :-)

MCUdude, 2022-01-28:
Yes, I know it sucks when you have to redesign because of a tiny mistake. Good luck with your design!

closed 2022-01-28 by MCUdude