MCUdude/MightyCore #238: Suggestion - JTAG support

bb5317, 2022-01-31:
Please consider adding a JTAG enable/disable option to the boards file so the proper fuse setting is programmed. Some designs do not use Serial and JTAG is the only way to program the device. The default fuse setting disables JTAG which effectively bricks a MCU that is already soldered to the PCB.

bb5317, 2022-01-31:
I understand. I wasn't looking to learn a new IDE as the Arduino core is good enough for a quick piece of code. I was looking for a way to not have to modify the boards file every time there is an update to your core. I've attached modified boards files for MightyCore 213 and MegaCore 213.
[boards - JTAG MegaCore213.txt](https://github.com/MCUdude/MightyCore/files/7974219/boards.-.JTAG.MegaCore213.txt)
[boards - JTAG MightyCore213.txt](https://github.com/MCUdude/MightyCore/files/7974220/boards.-.JTAG.MightyCore213.txt)

closed 2022-01-31 by bb5317

MCUdude, 2022-01-31:
Your best option would probably be to fork MegaCore and MightyCore, apply your modification, and install them "manually", by cloning/downloading them into your Arduino/hardware folder.

Both MightyCore and MegaCore are more or less in "maintenance mode", so you will probably not miss anything major if you sync your forks once a year.