MCUdude/MightyCore #240: Atmega8535 project

ITstreet1, 2022-02-08:
What pins I can use for Software Serial on this MCU? As far as I know, for RX it must be an interrupt pin. So it can be PD2, PD3, or PB2. For TX can be anyone. Am I right?

MCUdude, 2022-02-08:
> As far as I know, for RX it must be an interrupt pin. So it can be PD2, PD3, or PB2. For TX can be anyone. Am I right?

Correct!

closed 2022-04-24 by MCUdude

ITstreet1, 2022-05-06:
I made the board. The bootloader goes fine. I can upload a sketch, and blink an LED. 
But, when I try to upload a barebone example with only Wire.h included, I get this error:

> C:\Users\----\Documents\Arduino\libraries\Wire\Wire.cpp: In member function 'void TwoWire::begin()':
> C:\Users\----\Documents\Arduino\libraries\Wire\Wire.cpp:62:3: error: 'DIDR0' was not declared in this scope
>    DIDR0 &= 0xCF;
>    ^~~~~
> C:\Users\----\Documents\Arduino\libraries\Wire\Wire.cpp:62:3: note: suggested alternative: 'DDA0'
>    DIDR0 &= 0xCF;
>    ^~~~~
>    DDA0
> Multiple libraries were found for "Wire.h"
>  Used: C:\Users\----\Documents\Arduino\libraries\Wire
>  Not used: C:\Users\----\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.3\libraries\Wire
> Multiple libraries were found for "SPI.h"
>  Used: C:\Users\----\Documents\Arduino\libraries\SPI
>  Not used: C:\Users\----\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.3\libraries\SPI
> exit status 1
> Error compiling for board ATmega8535.

MCUdude, 2022-05-06:
Your error message clearly shows the issue:

> Multiple libraries were found for "Wire.h"
Used: C:\Users----\Documents\Arduino\libraries\Wire
Not used: C:\Users----\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.3\libraries\Wire
Multiple libraries were found for "SPI.h"
Used: C:\Users----\Documents\Arduino\libraries\SPI
Not used: C:\Users----\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.3\libraries\SPI
exit status 1

You will have to either rename or remove the SPI and Wire libraries located in C:\Users----\Documents\Arduino\libraries. These are some custom ones you have installed that interferes with the built-in ones.

ITstreet1, 2022-05-06:
I missed that. 
But this error never came up when compiling for other MCUs. Say ESP8266, ESP32, Uno, etc. The reason I didn't even notice this error as I should. This lib is there for a while, never making problems.

MCUdude, 2022-05-06:
> But this error never came up when compiling for other MCUs. Say ESP8266, ESP32, Uno, etc.

It may be because it has `architectures=avr` in the library.properties file. If the library is written for a UNO, there's your problem. The ATmega8535 is also an AVR, but the library has not been developed for it specifically.