MCUdude/MightyCore #241: Proteus Simulation Clock

5arid, 2022-02-15:
I am trying to simulate a simple blink led in proteus by using MightyCore but the led blinks too faster than 1s.
Here is what I do in summury :
Arduino --> MightyCore --> ATmega16 --> Internal 8MHz --> export compiled binary
Proteus --> ATmega16 --> Fuse : Int RC 8MHz --> Load hex file
I noticed that if I set the clock to 0.08MHz (in proteus) it seems to work!
I can't find out if this issue is with Proteus or MightyCore or Arduino IDE itself!
Also MiniCore and ATmega8 works fine but what is wrong with MightyCore and ATmega16?

MCUdude, 2022-02-15:
I have never used Proteus, so I have no idea. It works perfectly on real hardware, and that's really all I care about

closed 2022-04-24 by MCUdude

rtek1000, 2023-05-20:
> I am trying to simulate a simple blink led in proteus by using MightyCore but the led blinks too faster than 1s. Here is what I do in summury : Arduino --> MightyCore --> ATmega16 --> Internal 8MHz [...]

If someone else tries to simulate with this core, be warned that in the simulation the MCU Atmega1284, Atmega644, Atmega324 and Atmega164 didn't even oscillate the pin to flash the LED.

rtek1000, 2023-05-20:
> If someone else tries to simulate with this core, be warned that in the simulation the MCU Atmega1284, Atmega644, Atmega324 and Atmega164 didn't even oscillate the pin to flash the LED.

I managed to make the Atmega1284 (I also tested with Atmega16, and it worked. The others I haven't even tried, they certainly work with MPLAX and XC8.) work using MPLAX with XC8, so it must be a problem in the Arduino core. Perhaps some setting of the delay/clock functions need to be revised.

[ATmega16_blink.X.zip](https://github.com/MCUdude/MightyCore/files/11521368/ATmega16_blink.X.zip)
[Atmega1284_blink.X.zip](https://github.com/MCUdude/MightyCore/files/11521369/Atmega1284_blink.X.zip)