MCUdude/MightyCore #243: Error compiling for board ATmega1284

brown5500, 2022-05-08:
I have an ender 3 with a V1.1.3 board.  I downloaded the 1.1.x marlin files, and copied over all the example files from the Ender-3 folder.  I followed every instruction that I could find, but I'm getting an error compiling no matter what I try.  I got the same results on Arduino IDE 1.8.19 & 1.8.18.  I also tried a few different versions of the firmware.   I'm pasting the error messages below.  Thanks for your help!

Arduino: 1.8.18 (Windows 10), Board: "ATmega1284, Yes (UART0), Standard pinout, EEPROM retained, 1284P, BOD 2.7V, LTO disabled, External 16 MHz"

In file included from C:\Users\corya\Downloads\Marlin-1.1.x\Marlin-1.1.x\Marlin\HAL.h:36:0,

                 from C:\Users\corya\Downloads\Marlin-1.1.x\Marlin-1.1.x\Marlin\MarlinConfig.h:39,

                 from C:\Users\corya\Downloads\Marlin-1.1.x\Marlin-1.1.x\Marlin\enum.h:26,

                 from C:\Users\corya\Downloads\Marlin-1.1.x\Marlin-1.1.x\Marlin\planner.h:36,

                 from C:\Users\corya\Downloads\Marlin-1.1.x\Marlin-1.1.x\Marlin\planner.cpp:65:

C:\Users\corya\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.3\cores\MCUdude_corefiles/Arduino.h:125:22: error: statement-expressions are not allowed outside functions nor in template-argument lists

 #define sq(x)        ({ typeof (x) _x = (x); _x * _x; })

                      ^

C:\Users\corya\Downloads\Marlin-1.1.x\Marlin-1.1.x\Marlin\macros.h:95:32: note: in expansion of macro 'sq'

 #define CIRCLE_AREA(R) (M_PI * sq(float(R)))

                                ^~

C:\Users\corya\Downloads\Marlin-1.1.x\Marlin-1.1.x\Marlin\planner.cpp:153:44: note: in expansion of macro 'CIRCLE_AREA'

         Planner::volumetric_area_nominal = CIRCLE_AREA(float(DEFAULT_NOMINAL_FILAMENT_DIA) * 0.5f), // Nominal cross-sectional area

                                            ^~~~~~~~~~~

exit status 1

Error compiling for board ATmega1284.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

MCUdude, 2022-05-08:
What happens if you add the following code all the way on the top of the sketch:

``` c
#ifdef sq
  #undef sq
  #define sq(x) ((x)*(x))
#endif
```

brown5500, 2022-05-08:
Thanks for the suggestion.  I got the exact same result with that code
added though.  Just to make sure that I did it right, there are multiple
tabs in the Arduino setup.  The first one is Marlin. The top of that one is
where I added your code.

MCUdude, 2023-07-25:
The macros have later been replaced with template functions, and the issue should now be resolved.

closed 2023-07-25 by MCUdude