MCUdude/MightyCore #250: atmega32-16u Bootloader works once then fail after sketch flash

Necromonger2, 2022-08-10:
I am having an issue where the bootloader fails after loading a sketch via Arduino IDE.

Currently using a Arduino uno with ArduinoISP sketch loaded to flash the bootloader.
After flashing the bootloader i remove the ICSP cable and flash the blink sketch via FTDI.
When i try to flash another sketch i get the following error:

avrdude: Device signature = 0x1e950f
avrdude: Expected signature for ATmega32 is 1E 95 02

or

avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check
         
or

         Using Port                    : COM7
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x74
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xe0

avrdude done.  Thank you.

I have this issue with 8Mhz internal clock and 16Mhz external
I have also tried with atmega328p with the same results (the atmega328p works perfectly in the uno board)

Necromonger2, 2022-08-11:
Attached is the bootloader log then the first sketch flash (Successful) and then the failed 2nd flash
For reference the board settings
[image]
[bootloader_log.txt](https://github.com/MCUdude/MightyCore/files/9310440/bootloader_log.txt)
[FirtsFlash.txt](https://github.com/MCUdude/MightyCore/files/9310441/FirtsFlash.txt)
[SecondFlash.txt](https://github.com/MCUdude/MightyCore/files/9310442/SecondFlash.txt)

JAndrassy, 2022-08-11:
do you reset it right before the upload?

MCUdude, 2022-08-11:
When looking at the second flash, it looks like you've built and trying to upload with the clock set to "8 MHz internal" instead of "20 MHz external".
```
fqbn=MightyCore-master:avr:32:bootloader=uart0,eeprom=keep,pinout=standard,BOD=2v7,LTO=Os,clock=8MHz_internal -vid-pid=067B_2303
```

Make sure you don't change the clock settings after you've burned the bootloader. Select one and stick with it

Necromonger2, 2022-08-12:
my apologies i did notice the clock change and i did set it to 16mhz. I do
reset the mcu after bootloader flash. i usually pull out the supply so that
there is no power and until all led's turn off then only do i plug the
supply back in. I also did try with just pressing the reset button. results
remain the same

rocifier, 2022-09-28:
Yeah I did this today too but it somehow burned all 0's onto the fuses. I ignored the safety warning to revert the fuses because I didn't understand it. And it bricked my board because I don't have an external crystal. Luckily I got 5 boards made incase this happened. Good to know to "pick one and stick with it" as well.

mcuee, 2023-05-19:
I think the issue can be closed.

Necromonger2, 2023-05-19:
I have not yet found a solution to the problem. Also haven't worked on an
atmega32-16u since

MCUdude, 2023-05-19:
>  I do reset the mcu after bootloader flash. i usually pull out the supply so that
there is no power and until all led's turn off then only do i plug the
supply back in. I also did try with just pressing the reset button. results remain the same

The problem is that you don't have an auto reset circuitly that works properly. If you want to use the bootloader, you should absolutely have the auto reset circuity present. If not, you'll have to reset your board at the right time, and this can be very difficult. When you apply power to your board, the bootloader detectrs that a power-on reset occured, instead of the RESET pin being pulled low, and it will just straight into the sketch without staying in bootloader mode.

The fact that you can upload using the bootloader only once is a clear sign of this. When the bootloader is "fresh", and there is no sketch in the flash memory, the bootloader will start over and over again.

This is clearly a user/hardware issue, so I'm closing this issue. The Arduino forum is a great place if you need further help getting your hardware working.

closed 2023-05-19 by MCUdude