MCUdude/MightyCore #252: Menu-option for turning off JTAG interface

fschimpf, 2022-09-26:
Hi, I am using MightyCory on an atmega644a and realized that the JTAG-interface is always on and thus blocking pins 19, 20, 21 from being used as general digitalIO. Would it be possible to add an option in the menu for turning off JTAG (JTAGEN, hfuse, bit 6) for making the pins usable?
Thanks for the good work on MithyCore!
Fritz

fschimpf, 2022-09-26:
Maybe I should mention that I am not using the bootloader. It seems like the fuses were still on their defaults, with enabled JTAG. I am setting the fuses manually after uploading of the program as a workaround now.

MCUdude, 2022-09-26:
the JTAGEN fuse is enabled by default on factory fresh AVRs. If you select "No Bootloader" in the Arduino IDE tools menu, and then click "Burn bootloader", you're only setting the fuses, which disables the JTAG interface.

3rd party Arduino cores like MightyCore can't change the label on the "Burn Bootloader" button, even though the settings in the tools menu decides what that button should do.

fschimpf, 2022-09-26:
Cool, thanks!

closed 2022-09-26 by MCUdude