MCUdude/MightyCore #254: compile error...

BNSF-FAN, 2022-10-11:
I cannot compile a sketch, I will post all errors(?), maybe I'm doing something wrong. No forums will help me saying they don't know about fixing the issue. I don't know where to go. I am not having any luck. Can you help me out?

Here is what the IDE gave me; 

Arduino: 1.8.19 (Windows 7), Board: "ATmega32, No bootloader, EEPROM retained, Standard pinout, BOD disabled, LTO disabled, Internal 1 MHz"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Kims\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Kims\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Kims\Documents\Arduino\libraries -fqbn=MightyCore:avr:32:bootloader=no_bootloader,eeprom=keep,pinout=standard,BOD=disabled,LTO=Os,clock=1MHz_internal -vid-pid=2341_0043 -ide-version=10819 -build-path C:\Users\Kims\AppData\Local\Temp\arduino_build_363945 -warnings=all -build-cache C:\Users\Kims\AppData\Local\Temp\arduino_cache_619963 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\Kims\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=C:\Users\Kims\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avr-gcc.path=C:\Users\Kims\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\Kims\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Kims\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\Kims\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose I:\1A CTC SIGNAL SCHEMATICS\CONTROL POINT FIRMWARE\RIGHT HAND TURNOUT CONTROL POINT\RIGHT_HAND_TURNOUT_CONTROL_POINT_Firmware\RIGHT_HAND_TURNOUT_CONTROL_POINT_Firmware.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Kims\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Kims\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Kims\Documents\Arduino\libraries -fqbn=MightyCore:avr:32:bootloader=no_bootloader,eeprom=keep,pinout=standard,BOD=disabled,LTO=Os,clock=1MHz_internal -vid-pid=2341_0043 -ide-version=10819 -build-path C:\Users\Kims\AppData\Local\Temp\arduino_build_363945 -warnings=all -build-cache C:\Users\Kims\AppData\Local\Temp\arduino_cache_619963 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\Kims\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=C:\Users\Kims\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avr-gcc.path=C:\Users\Kims\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\Kims\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Kims\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\Kims\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose I:\1A CTC SIGNAL SCHEMATICS\CONTROL POINT FIRMWARE\RIGHT HAND TURNOUT CONTROL POINT\RIGHT_HAND_TURNOUT_CONTROL_POINT_Firmware\RIGHT_HAND_TURNOUT_CONTROL_POINT_Firmware.ino

Using board '32' from platform in folder: C:\Users\Kims\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.3

Using core 'MCUdude_corefiles' from platform in folder: C:\Users\Kims\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.3

Warning: Board arduino:avr:atmega32o doesn't define a 'build.board' preference. Auto-set to: AVR_ATMEGA32O

Warning: Board arduino:avr:atmega16o doesn't define a 'build.board' preference. Auto-set to: AVR_ATMEGA16O

Warning: Board arduino:avr:atmega644o doesn't define a 'build.board' preference. Auto-set to: AVR_ATMEGA644O

Warning: Board arduino:avr:atmega1284o doesn't define a 'build.board' preference. Auto-set to: AVR_ATMEGA1284O

Detecting libraries used...

"C:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega32 -DF_CPU=1000000L -DARDUINO=10819 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.1.3\\variants\\standard" "C:\\Users\\Kims\\AppData\\Local\\Temp\\arduino_build_363945\\sketch\\RIGHT_HAND_TURNOUT_CONTROL_POINT_Firmware.ino.cpp" -o nul

Alternatives for ServoTimers.h: [Servo@1.1.1]

ResolveLibrary(ServoTimers.h)

  -> candidates: [Servo@1.1.1]

"C:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega32 -DF_CPU=1000000L -DARDUINO=10819 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.1.3\\variants\\standard" "-IC:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.1.3\\libraries\\Servo\\src" "C:\\Users\\Kims\\AppData\\Local\\Temp\\arduino_build_363945\\sketch\\RIGHT_HAND_TURNOUT_CONTROL_POINT_Firmware.ino.cpp" -o nul

"C:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega32 -DF_CPU=1000000L -DARDUINO=10819 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.1.3\\variants\\standard" "-IC:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.1.3\\libraries\\Servo\\src" "C:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.1.3\\libraries\\Servo\\src\\Servo.cpp" -o nul

Generating function prototypes...

"C:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega32 -DF_CPU=1000000L -DARDUINO=10819 -DARDUINO_AVR_ATmega32 -DARDUINO_ARCH_AVR "-IC:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.1.3\\variants\\standard" "-IC:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\MightyCore\\hardware\\avr\\2.1.3\\libraries\\Servo\\src" "C:\\Users\\Kims\\AppData\\Local\\Temp\\arduino_build_363945\\sketch\\RIGHT_HAND_TURNOUT_CONTROL_POINT_Firmware.ino.cpp" -o "C:\\Users\\Kims\\AppData\\Local\\Temp\\arduino_build_363945\\preproc\\ctags_target_for_gcc_minus_e.cpp"

"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Kims\\AppData\\Local\\Temp\\arduino_build_363945\\preproc\\ctags_target_for_gcc_minus_e.cpp"

Compiling sketch...

"C:\\Users\\Kims\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32 -D [... cut ...]
C:\Users\Kims\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.3\libraries\Servo\src/ServoTimers.h:65:41: note: previous declaration as 'typedef enum timer16_Sequence_t timer16_Sequence_t'

 typedef enum { _timer1, _Nbr_16timers } timer16_Sequence_t ;

                                         ^~~~~~~~~~~~~~~~~~

Using library Servo at version 1.1.1 in folder: C:\Users\Kims\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.3\libraries\Servo 

exit status 1

Error compiling for board ATmega32.

Please, can you help me? I am really confused...

Many, many thanks!!

Kim

MCUdude, 2022-11-08:
Can you post your sketch here? It looks like the Servo library is conflicting with another library.

oqibidipo, 2022-11-25:
`ServoTimers.h` does not have include guards and it is getting included twice, first directly (and unnecessarily) from sketch line 2 and the from `Servo.h` which is included from sketch line 4. 

```
In file included from libraries\Servo\src/Servo.h:52:0,
                 from RIGHT_HAND_TURNOUT_CONTROL_POINT_Firmware.ino:4:
libraries\Servo\src/ServoTimers.h:65:16: error: redeclaration of '_timer1'

In file included RIGHT_HAND_TURNOUT_CONTROL_POINT_Firmware.ino:2:0:
libraries\Servo\src/ServoTimers.h:65:16: note: previous declaration 'timer16_Sequence_t _timer1'

----

In file included from libraries\Servo\src/Servo.h:52:0,
                 from RIGHT_HAND_TURNOUT_CONTROL_POINT_Firmware.ino:4:
libraries\Servo\src/ServoTimers.h:65:25: error: redeclaration of '_Nbr_16timers'

In file included from RIGHT_HAND_TURNOUT_CONTROL_POINT_Firmware.ino:2:0:
libraries\Servo\src/ServoTimers.h:65:25: note: previous declaration 'timer16_Sequence_t _Nbr_16timers'

----

In file included from libraries\Servo\src/Servo.h:52:0,
                 from RIGHT_HAND_TURNOUT_CONTROL_POINT_Firmware.ino:4:
libraries\Servo\src/ServoTimers.h:65:41: error: conflicting declaration 'typedef enum timer16_Sequence_t timer16_Sequence_t'

In file included from RIGHT_HAND_TURNOUT_CONTROL_POINT_Firmware.ino:2:0:
libraries\Servo\src/ServoTimers.h:65:41: note: previous declaration as 'typedef enum timer16_Sequence_t timer16_Sequence_t'
```

@BNSF-FAN 
Remove `#include <ServoTimers.h>`.

@MCUdude 
Add include guards to `ServoTimers.h`.

commit 6e4649ee (2023-01-04): Add include guards
Closes #254
closed 2023-01-04 by MCUdude