MCUdude/MightyCore #256: Upload once successfully but can't after that

unebonnevie, 2022-11-13:
Hi,

I followed the installation instructions on the MigthCore github site for my ATMEGA644 using the standard pin config.  See image for the setup. I flash the bootloader successfully.

Loaded up the Blink example and uploaded it successfullly.  See image below.  Now attempted to upload the same Blink sketch again, and I get an error.  See image below.  To do it successfully, I would have burn the bootloader again, and it will give me a one-time successful upload.

Please advise!

[image]

[image]

[image]

unebonnevie, 2022-11-13:
Here is my AVRDude command:

`C:\Users\me\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude -CC:\Users\me\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.3/avrdude.conf -v -patmega644p -carduino -PCOM3 -b115200 -D -Uflash:w:C:\Users\asle\AppData\Local\Temp\arduino_build_314197/Blink.ino.hex:i 

`

MCUdude, 2022-11-14:
> but the bootloader is 65536 bytes in size, filling up the ATMEGA644 flash space.

This is not correct, the bootloader size is a little more than 512 bytes large. Avrdude reports 65536 bytes because the bootloader is located at the very end of the flash memory. If you dump the flash memory (using Avrdude directly), you can see that the bootloader takes up very little space in the flash memory.

If uploading only works once, it's _highly likely_ that your auto-reset circuit isn't present or working properly. 
Have you connected a 100nF capacitor between the DTR or RTS line on your USB to serial adapter and to the RESET pin on the ATmega644P? And do you have a 10k pullup on the reset pin? If you have a schematic, that would be helpful.

unebonnevie, 2022-11-19:
>Have you connected a 100nF capacitor between the DTR or RTS line on your USB to serial adapter and to the RESET pin on the ATmega644P?

Thank you!  That was the problem!  I saw your documentation, but didn't go for the 0.1uF to the USB-serial chip's DTR pin and the MCU's RESET pin, because I thought it was OK, since it was able to flash the very first time.

With 0.1uF connected, I am able to flash through the bootloader.

closed 2022-11-29 by MCUdude