MCUdude/MightyCore #257: Serial print output is scrambled

garudaonekh, 2022-11-28:
Hi,
I have atmega16. I burn bootloader with 16MHz internal.
I use TX0,RX0 for printing serial output. But all the output are unreadable. I tried several baudrates, but still the same. I also try to use 3.3v but still the same.

I used to use MiniCore with atmega168/atmega88, all the serial print work fine.

thanks;

MCUdude, 2022-11-28:
> I have atmega16. I burn bootloader with 16MHz internal.

There's no such thing as a 16 MHz internal oscillator. Which clock speed are you running, and are you using an internal or external oscillator?

And are you able to upload using the bootloader just fine?

garudaonekh, 2022-11-28:
> > I have atmega16. I burn bootloader with 16MHz internal.
> 
> There's no such thing as a 16 MHz internal oscillator. Which clock speed are you running, and are you using an internal or external oscillator?
> 
> And are you able to upload using the bootloader just fine?

Sorry, Internal 8Mhz 

Yes, I can upload the code and it works because I see the serial output but they are all unreadable.

MCUdude, 2022-11-28:
Which baud rate are you using when the print output is scrambled?

It would be great if you could post a screenshot of your settings in the IDE tools menu and post the Avrdude output when you upload. Make sure to turn on verbose upload output in the IDE settings.

garudaonekh, 2022-11-28:
Now it's ok. I reconnect the tx/rx again and change the USB.

Thanks;

closed 2022-11-28 by garudaonekh