MCUdude/MightyCore #258: Remote Control using Atmega + Lora Chip

garudaonekh, 2022-11-29:
Hi,
I want to make remote control with 8 buttons(Atmega16+LLCC68 Lora) with 9v battery to control 4 relays(4 buttons to turn on and 4 off respectively). The relays board will immediately send acknowledgement back to the remote control. Battery life is a concern here.

**My idea:**
I plan to use pins interrupt to wakeup Atmega16 when buttons are pressed and the Atmega will go to sleep after buttons are pressed and processing incoming acknowledgement. But I find out that Atmega16 has only 3 wakeup interrupt pins(correct me if I am wrong). 

Please share your suggestion.

**Where I am now:**
So far I manage send data and receive data using LLCC68+Atmega16a.

MCUdude, 2022-11-29:
If the four buttons are pulled low when not pressed and go high when pressed, you can use four diodes and create a single wakeup pin for all buttons. Alternatively, you can do this for the last two buttons.

Like this. I've not tested it, but I believe it should work. use pinMode(pin, INPUT) instead of INPUT_PULLUP to prevent power usage, since the WAKEUP and BUTTON signals are pulled low by default.

[image]

garudaonekh, 2022-12-08:
Thanks, it works.

closed 2022-12-08 by garudaonekh