MCUdude/MightyCore #261: Problems with external strong (4-pin) oscillator

TheByteAttic, 2023-01-02:
Hi, this is a great and extremely useful project.
I want to use the ATmega1284P with a strong, 4-pin external oscillator (thus no crystal or capacitors) at 16MHz, feeding into pin 13 (XTAL1). So I changed the boards.txt file in my local installation to use the following configuration for the low fuses: 0xF0. Burning the bootloader with an Arduino Uno reports success, but the LED on PB0 does not blink (and indeed, PB0 does not change potential).
Nonetheless, I then hooked up an FTDI adapter to try to upload the blinking sketch with the AVR ISP (MightyCore) programmer , but it fails with the following error:
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x4f
Trying other programmers doesn't seem to help.
What am I doing wrong?
Cheers, Bernardo.

closed 2023-01-02 by TheByteAttic

MCUdude, 2023-01-02:
Did you figure out the issue?

TheByteAttic, 2023-01-02:
Hi, yes, apologies, it was a silly mistake. I wanted to delete the issue but didn't figure out how to do it, so I closed it. Thanks for getting back to me nonetheless!