MCUdude/MightyCore #262: ATMEGA324PA bootloader upload error

barcesat, 2023-01-02:
Hi, thanks for the great work and documentation!
I created a test board with ATMEGA324PA in two variants (8MHz & 16MHz external crystals). 
Here's the config inthe IDE for the 8MHz variant:
[image]

I've been trying to burn bootloader to them using an Atmel ICE but had the following errors:

```
avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\barce\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.1.3/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : atmelice_isp
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega324PA
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       1024    4      0  9000  9000 0xff 0xff
           flash         33     6   256    0 yes     32768  128    256  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : JTAG3_ISP
         Description     : Atmel-ICE (ARM/AVR) in ISP mode
         Vtarget         : 4.3 V
         SCK period      : 8.00 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9511 (probably m324pa)
avrdude: erasing chip
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: WARNING: invalid value for unused bits in fuse "lock", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0xff instead of 0x3f (double check with your datasheet first).
avrdude: 1 bytes of lock verified
avrdude: reading input file "0b11111101"
avrdude: writing efuse (1 bytes):

Writing | ################################################## | 100% 0.09s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0b11111101:
avrdude: load data efuse data from input file 0b11111101:
avrdude: input file 0b11111101 contains 1 bytes
avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xff != 0xfd
avrdude: verification error; content mismatch

avrdude done.  Thank you.

Error while burning bootloader.
```

I'm using Arduino IDE 1.8.15, MightyCore 2.1.3.
I'm able to upload the blink example ith the programmer (both the C one and the arduino one).
What am I missing in order to burn the bootloader correctly?

I have another issue with the timings on the 16MHz varant that seems to get the timing kinda right only when I adjust the delay time to 100 instead of 1000. I will test this with a scope to confirm and update.

mcuee, 2023-02-10:
Try upgrade to the latest MightyCore 2.2.1 version which uses avrdude 7.1 version to see if that help.

MCUdude, 2023-07-25:
Issue most likely resolved with the latest MightyCore version (using Avrdude 7.1)

closed 2023-07-25 by MCUdude