MCUdude/MightyCore #267: Atmega32-16u get error use encoder library

El-Rafa, 2023-03-03:
please help how to fix it :

In file included from C:\Users\El Rafa\Documents\Arduino\libraries\Encoder-master/Encoder.h:41:0,
                 from C:\Users\El Rafa\Documents\Arduino\sketch_mar03a\sketch_mar03a.ino:2:
C:\Users\El Rafa\Documents\Arduino\libraries\Encoder-master/utility/direct_pin_read.h:1:0: error: unterminated #ifndef
 #ifndef direct_pin_read_h_
 
In file included from C:\Users\El Rafa\Documents\Arduino\libraries\Encoder-master/Encoder.h:46:0,
                 from C:\Users\El Rafa\Documents\Arduino\sketch_mar03a\sketch_mar03a.ino:2:
C:\Users\El Rafa\Documents\Arduino\libraries\Encoder-master/utility/interrupt_pins.h:398:2: error: #error "Interrupts are unknown for this board, please add to this code"
 #error "Interrupts are unknown for this board, please add to this code"
  ^~~~~
C:\Users\El Rafa\Documents\Arduino\libraries\Encoder-master/utility/interrupt_pins.h:401:2: error: #error "Encoder requires interrupt pins, but this board does not have any :("
 #error "Encoder requires interrupt pins, but this board does not have any :("
  ^~~~~
C:\Users\El Rafa\Documents\Arduino\libraries\Encoder-master/utility/interrupt_pins.h:402:2: error: #error "You could try defining ENCODER_DO_NOT_USE_INTERRUPTS as a kludge."
 #error "You could try defining ENCODER_DO_NOT_USE_INTERRUPTS as a kludge."
  ^~~~~
Using library Encoder-master at version 1.4.2 in folder: C:\Users\El Rafa\Documents\Arduino\libraries\Encoder-master 
exit status 1
Error compiling for board ATmega32.

MCUdude, 2023-03-03:
I'll assume this is the library you're using.

https://github.com/PaulStoffregen/Encoder/issues

The ATmega16 is probably not supported out of the box, but you should ask the library author just in case.

El-Rafa, 2023-03-03:
> I'll assume this is the library you're using.
> 
> https://github.com/PaulStoffregen/Encoder/issues
> 
> The ATmega16 is probably not supported out of the box, but you should ask the library author just in case.

yes, i use this encoder library.
thank for your explanation, i'll ask to the library author.

closed 2023-05-01 by MCUdude