MCUdude/MightyCore #268: Can't update boards.txt

TheByteAttic, 2023-03-04:
Since updating the Arduino IDE to v2.0.4, whatever I change in the file boards.txt (such as changing the fuse values or even the menu item's name) is ignored. I've tried to do it both in my sketchbook folder:
C:\Users\MyName\OneDrive\Documents\Arduino\hardware\MightyCore-master\avr\boards.txt
as well as in the installation folder"
C:\Users\MyName\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.2.1\boards.txt
Whatever I edit in either of these files is ignored when I open the IDE and go to boards or clock options. I already tried reinstalling the MightyCore library and no cigar. Do you recognize the issue?

TheByteAttic, 2023-03-04:
Issue solved by installing older version 2.1.3 of MightyCore.

MCUdude, 2023-03-06:
IIRC the IDE 2.0 still has an issue where the tools menu option for a "Manually installed" core doesn't appear. That's why using the boards manager for installation is recommended. It also downloads all the assets like Avrdude 7.1, but with a manual install, this has to be handled manually.

However, I have no issues editing the boards.txt found in the installation folder on my computer:
(/Users/hans/Library/Arduino15/packages/MightyCore/hardware/avr/2.2.1/boards.txt)

[image]

mcuee, 2023-05-19:
I think the issue can be closed.

closed 2023-05-19 by MCUdude