MCUdude/MightyCore #269: Custom baud?

jedahan, 2023-04-03:
I have a melzi board, which contains an atmega1284p, but is running a bootloader that expects to upload at 57600 baud.

Is there a way to extend the current menu options to include customizing the programmer baud rate? Currently using 'arduino as ISP'.

Would this be a welcome PR? I can dig a bit and see how to add menu options.

MCUdude, 2023-04-03:
> I have a melzi board, which contains an atmega1284p, but is running a bootloader that expects to upload at 57600 baud.

If you replace the bootloader, you can have it run at 115200 baud, which is what is most common for AVRs running at 16 MHz.

It's not as easy as adding a baud rate menu option. The boards.txt file and its parser is very primitive, and one would end up with lots of baud rates that isn't compatible with the selected clock speed.

If you want to upload at 57600 baud and compile for a 16 MHz clock, I'll recommend you to use PlatformIO instead, where settings can be specified in a project configuration file.

mcuee, 2023-05-12:
Or just plain avrdude command line to upload.

closed 2023-07-25 by MCUdude