MCUdude/MightyCore #273: Perhaps some setting of the delay/clock functions need to be revised.

rtek1000, 2023-05-20:
In the simulation the MCU Atmega1284, Atmega644, Atmega324 and Atmega164 didn't even oscillate the pin to flash the LED.

I managed to make the Atmega1284 work using MPLAX and XC8.

https://github.com/MCUdude/MightyCore/issues/241

[ATmega16_blink.X.zip](https://github.com/MCUdude/MightyCore/files/11521375/ATmega16_blink.X.zip)
[Atmega1284_blink.X.zip](https://github.com/MCUdude/MightyCore/files/11521376/Atmega1284_blink.X.zip)

MCUdude, 2023-07-25:
See reply in #241. I have no idea why the Proteus doesn't with MightyCore, but you're welcome so submit a PR if you think you know how to resolve the issue

closed 2023-07-25 by MCUdude

rtek1000, 2023-07-25:
Thank you for your attention. I believe that other users also stopped using this core because it was not possible to simulate the circuit. I hope someone can resolve this in the future.