MCUdude/MightyCore #274: OC1b vs XCK1, OC2A vs XCK2

RobissX, 2023-05-21:
Hi, 

I have the Problem that i can't use PWM(OC1B PD4) and Serial**1** at the same time.
If i have pwm on the output pd4 it works normally until i use Serial1 then pwm signal goes away(Normal PORTD HIGH LOW works).
If i disable Serial1 then the PWM signal come back.
and all the same Problem for OC2A and Serial2

I've already checked that Serial is always in asynchronous mode!

More info:
https://forum.arduino.cc/t/atmega324pb-pwm-hate-serial-oc1b-vs-xck1-oc2a-vs-xck2

MCUdude, 2023-05-21:
Huh, that's strange! Sounds almost like a silicon bug.

What happens if you do the following (in this exact order)
* Initialize Serial1
* Set pin PD4 as input
* enable a PWM signal on pin PD4

RobissX, 2023-05-22:
I've tried but without success.
I tried the same code with an atmega324pa and everything works normally there. I even uploaded the compile code for the atmega324pa to the atmega324pb without success. 

Really looks like a silicon bug.🙁

The only thing that has changed in the Atemga324pb is "UCSRnD" and "USART0 Start Frame detection" Interrupt.

Does anyone have contact to Atmel/Microchip?

MCUdude, 2023-05-22:
Could you try writing a bare-bone example and ask over at the Avrfreaks forum? Those guys are true experts and even some Microchip developers are watching the forum. You can still use Arduino IDE to compile your code. Just get rid of `void setup()` and `void loop()`, and use `int main()` instead

RobissX, 2023-05-22:
You mean like that
```
int main() {
  DDRD |= (1 << PD4);
  TCCR1A |= (1 << COM1B1) |  (1 << WGM10); //8bit Phase Correct
  TCCR1B |= (1 << CS11) | (1 << CS10); //64 prescaler

  OCR1B = 127;

  UCSR1A = 1;
  UCSR1C = 6;

  while (1) {
    UCSR1B = 152;
    _delay_ms(1000);
    UCSR1B = 0;
    _delay_ms(1000);
  }
}
```

Compiled for Atmega324pb->

Upload to Atmega324pb:
PWM, no PWM, PWM, no PWM.

Upload to Atmega324pa same hex:
PWM continuous.

Compiled for Atmega324pa->

Upload to Atmega324pa:
PWM continuous.

Upload to Atmega324pb same hex:
PWM, no PWM, PWM, no PWM.

I found on something on Avrfreaks but with no solution.
https://www.avrfreaks.net/s/topic/a5C3l000000UbS6EAK/t157758

MCUdude, 2023-07-25:
I'm closing this issue because there is nothing I/we can do about this, other than reporting it to Microchip and crossing our fingers that they will fix this silicon bug in the future. It's a real bummer though 😞

closed 2023-07-25 by MCUdude

RobissX, 2023-07-25:
I wrote with microchip and they admitted it's really a bug and they can't do anything.
They try to fix the bug with the next chip revision.