MCUdude/MightyCore #277: ATmega1284 clock source problem

lax1dude, 2023-06-24:
Hi, I'm trying to use MightyCore to program an ATmega1284 using an AVRISP MKII programmer. I have an external 20MHz crystal oscillator connected to XTAL1 and XTAL2 with two 20pF capacitors to ground (I tested 10pF as well) but when I try to upload the sketch with the "External 20MHz" option, the MCU doesn't appear to make use of the crystal, instead it seems like it uses an internal 1MHz clock source from somewhere else but still with all delay functions and such in the sketch compiled for a 20MHz clock, which causes all delay functions and millis and all that to run at 1/20 speed.

If I select "Internal 8MHz" I also end up with a chip that appears to only be running at 1MHz but with all delay functions and millis compiled for 8MHz and therefore running at 1/8 speed.

I've confirmed the ratios of 1/20 for 20MHz and 1/8 for the internal 8MHz using an oscilloscope by programming a sketch that turns a pin on and off every "16 milliseconds" and then measuring the period of the resulting signal with different clock speeds selected. A period of 16 milliseconds becomes exactly 320 milliseconds at 20MHz and becomes exactly 128 milliseconds at 8MHz.

I know it is using some internal fallback because if I remove the crystal entirely from the board the MCU still continues running like nothing happened even with "External 20MHz" clock and still at the same 1/20 speed. I've skimmed the ATmega1284 datasheet and I did not see anything about there being an automatic internal fallback clock if the external crystal fails, although I've seen something like it mentioned in the datasheets of other ATmega chips but I could be wrong. A system like that most likely would not explain why the internal clock options also are all defaulting to this 1MHz source too though.

Does anyone know what would cause this? And obviously, a solution other than just giving up and adjusting boards.txt to fix the time scaling and using the chip at only 5% its intended clock speed?

Sorry if this is a duplicate

lax1dude, 2023-06-24:
When I run this sketch:
```
void loop() {
  digitalWrite(1, HIGH);
  digitalWrite(1, LOW);
}
```
My oscilloscope measures the frequency at pin 1 as only about 8.25 kHz which makes no sense at all

lax1dude, 2023-06-25:
I tried programming it in Atmel studio and it worked perfectly both with internal and with the 20MHz crystal oscillator with the delay functions scaled correctly, if I had to guess MightyCore is not setting the required fuses for some reason.

And even if it did set the fuses that are in boards.txt correctly they did not turn out to be the fuses required for a "20MHz External" clock source. boards.txt says it uses `0xf7` as the low fuses for 20MHz external but Atmel studio sets the low fuses to `0xd7` for the clock source configuration that actually works with the 20MHz crystal I have so that probably also needs to be fixed.

AR-DEV-1, 2023-06-25:
lax1dude check this out. Eaglercraft 1.8.9 
GitHub Repo Link:
https://github.com/AR-DEV-1/Eaglercraft-1.8.9
Play:
https://ar-dev-1.github.io/Eaglercraft-1.8.9/

MCUdude, 2023-07-03:
In order to _just set the fuses_ using Arduino IDE, you'll have to click "Burn Bootloader", even though you're not actually flashing one.

By doing so (and selecting "No bootloader" in the bootloader menu option", you'll set the low fuse to 0xf7, which is correct. 

The only difference between 0xd7 and 0xf7 is the start-up time and delay.

Arduino IDE doesn't let 3rd party cores like MightyCore change the label of this button, so that's why it's a bit misleading.

closed 2023-07-03 by MCUdude

lax1dude, 2023-07-03:
okay thanks

lax1dude, 2023-07-05:
Hey, sorry for the late follow up I never got around to it after fully mentally processing your original response, in the multiple hours I spent poking around at this thing I did repeatedly try the burn bootloader option literally under the exact suspicion that it was the problem you stated and it still had no effect. And when I said the fuses were wrong it was because when I tried the setting fuses listed in boards.txt through atmel studio it bricked the chip and I had to use a signal generator as a clock source to recover it but I guess you could rule that out as user error or that there's some defect with my crystal and it doesn't work without a delay.