MCUdude/MightyCore #278: ATmega32A reset after upload.

TheEccentricGenius, 2023-07-23:
The 32A chip fails to reset automatically when the upload is done, A manual reset is required after upload.

MCUdude, 2023-07-25:
Are you using an ISP programmer or a bootloader? Sounds like a hardware or driver issue to me...

TheEccentricGenius, 2023-07-25:
I am using a serial chip CH340E on a breadboard with the correct bootloader for serial and 16MHZ external crystal burned.

The circuitry is a 10K pull down resistor and 100nF capacitor hooked up to dtr.

This all works perfectly on a ATmega328P but not on the 32A....Maybe the is an issue with the chip? they were sold as "seller refurbished" but work fine in all other aspects.

MCUdude, 2023-07-25:
I've used a CH340E on many boards and never had an issue with them. I buy them from LCSC.com, so I know these chips are from a trusted supplier. It may be the ATmega32 chip. If you measure the reset line with an oscilloscope you should be able to see the reset pulse appear. You _can_ try to swap the 100nf capacitor with a 1uF or even 10uF to see if it helps.

TheEccentricGenius, 2023-07-25:
I'll try that out tomorrow.... I find it a pain that this circuit works perfectly well with the 328P.
The 32A chip may be faulty however.

MCUdude, 2023-07-25:
It may be as simple as the "programming ended" reset pulse is slightly shorter than the start pulse. And since the reset capacitor is quite small, maybe the reset line isn't pulled all the way down.

However, since MightyCore uses the latest version of Avrdude, you can actually connect the reset pin directly to the DTR or RTS pin on your USB to serial adapter, given that both the adapter and the microcontroller is running at the same voltage.

TheEccentricGenius, 2023-07-26:
I figured out how to make reset work....Just connected a diode parallel with the resistor like the Arduino auto-reset circuit and bingo it works.

I sampled it with the oscilloscope it was a spike down(not completely) and then a smaller spike up.... It looked a lot nicer with the diode in.

Thanks so much for your help MCUdude.

closed 2023-07-26 by TheEccentricGenius