MCUdude/MightyCore #282: Keep failure to flash the optiboot loader to the ATMEGA128P

unebonnevie, 2023-10-15:
Hi,
I am using the ATMEGA1284P and have followed the instructions to get the bootloader flashed.   My 1284P is programmed with 0xCE as the low fuse byte and 0xD9 as the high fuse byte.  I have it use a 16Mhz external crystal.

I keep getting the below failure while flashing the bootloader.  It doesn ot matter what 16Mhz 1284P board choice I choose.

Please see the image attached.  It's trying to flash a bootloader of 131702 bytes.  That's big!

Thanks for any tip!

[image]

`
avrdude: Using SCK period of 10 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9705 (probably m1284p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: Using SCK period of 10 usec
avrdude: reading input file "C:\Users\asle\dev\arduino\hardware\mighty-1284p\avr/bootloaders/optiboot/optiboot_atmega1284p.hex"
avrdude: writing flash (131072 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 131072 bytes of flash written
avrdude: verifying flash memory against C:\Users\asle\dev\arduino\hardware\mighty-1284p\avr/bootloaders/optiboot/optiboot_atmega1284p.hex:
avrdude: load data flash data from input file C:\Users\asle\dev\arduino\hardware\mighty-1284p\avr/bootloaders/optiboot/optiboot_atmega1284p.hex:
avrdude: input file C:\Users\asle\dev\arduino\hardware\mighty-1284p\avr/bootloaders/optiboot/optiboot_atmega1284p.hex contains 131072 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x1fc00
         0xff != 0x11
avrdude: verification error; content mismatch

avrdude done.  Thank you.

Error while burning bootloader.

`

unebonnevie, 2023-10-15:
Although I get a verification error when flashing the bootloader (see original post), I am actually able to flash any firmware on my 1284P via the bootloader in using the Arduino IDE's upload menu choice.

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9705 (probably m1284p)
avrdude: reading input file "C:\Users\asle\AppData\Local\Temp\arduino_build_733264/listfiles.ino.hex"
avrdude: writing flash (11156 bytes):

Writing | ################################################## | 100% 0.55s

avrdude: 11156 bytes of flash written
avrdude: verifying flash memory against C:\Users\asle\AppData\Local\Temp\arduino_build_733264/listfiles.ino.hex:
avrdude: load data flash data from input file C:\Users\asle\AppData\Local\Temp\arduino_build_733264/listfiles.ino.hex:
avrdude: input file C:\Users\asle\AppData\Local\Temp\arduino_build_733264/listfiles.ino.hex contains 11156 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.26s

avrdude: verifying ...
avrdude: 11156 bytes of flash verified

avrdude done.  Thank you.

MCUdude, 2023-10-16:
But you aren't even using MightyCore!
_Mighty Mini 1284p @ 16MHz, Optiboot @ 1Mbps_ is from another Arduino core

unebonnevie, 2023-10-16:
Hi,

[image]

Also ready installed.  I am going to update.

[image]

After updating I now have a "MightyCore" menu item in the "Boards" menu.

[image]

MCUdude, 2023-10-19:
You'll probably have to switch USB drivers. Download [Zadig](https://zadig.akeo.ie) and switch to `libusbK` I think.

closed 2023-11-30 by MCUdude