MCUdude/MightyCore #290: 'TIMSK3' was not declared in this scope

rvoogd, 2024-02-08:
Hi 

For a project with the atmega324PB i'm getting the following error:
'TIMSK3' was not declared in this scope when using the Timer3 interrupt mask
 
I had it working before but I think an update of the Mightycore library has overwritten a change made to the board library... now i can't find what was changed and where to get it working again.
It must have something to do with the registrer definition for the 324PB... but where to add?

Thanks

MCUdude, 2024-02-08:
You'll have to at least paste the full error message you're getting before I can guess what went wrong.

rvoogd, 2024-02-08:
> You'll have to at least paste the full error message you're getting before I can guess what went wrong.

Apologies - beginners error.
Copied in the output from the Arduino IDE (version 2.3.0) below. Mightycore version is 3.0.1
using the atmega324PB 

Thanks in advance for your help

```
c:\users\rens\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\iom324p.h:748:0: note: this is the location of the previous definition
 #define RXEN0   4
 
In file included from C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\cores\MCUdude_corefiles/Arduino.h:393:0,
                 from G:\projects\speedometer\odometer.h:1,
                 from G:\projects\speedometer\speedometer.ino:56:
C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\variants\standard/pins_arduino.h:519:0: warning: "UDRIE0" redefined
   #define UDRIE0  UDRIE
 
In file included from c:\users\rens\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\io.h:250:0,
                 from c:\users\rens\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\eeprom.h:38,
                 from C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\libraries\EEPROM\src/EEPROM.h:25,
                 from G:\projects\speedometer\speedometer.ino:55:
c:\users\rens\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\iom324p.h:749:0: note: this is the location of the previous definition
 #define UDRIE0  5
 
In file included from C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\cores\MCUdude_corefiles/Arduino.h:393:0,
                 from G:\projects\speedometer\odometer.h:1,
                 from G:\projects\speedometer\speedometer.ino:56:
C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\variants\standard/pins_arduino.h:520:0: warning: "TXCIE0" redefined
   #define TXCIE0  TXCIE
 
In file included from c:\users\rens\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\io.h:250:0,
                 from c:\users\rens\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\eeprom.h:38,
                 from C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\libraries\EEPROM\src/EEPROM.h:25,
                 from G:\projects\speedometer\speedometer.ino:55:
c:\users\rens\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\iom324p.h:750:0: note: this is the location of the previous definition
 #define TXCIE0  6
 
In file included from C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\cores\MCUdude_corefiles/Arduino.h:393:0,
                 from G:\projects\speedometer\odometer.h:1,
                 from G:\projects\speedometer\speedometer.ino:56:
C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\variants\standard/pins_arduino.h:521:0: warning: "RXCIE0" redefined
   #define RXCIE0  RXCIE
 
In file included from c:\users\rens\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\io.h:250:0,
                 from c:\users\rens\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\eeprom.h:38,
                 from C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\libraries\EEPROM\src/EEPROM.h:25,
                 from G:\projects\speedometer\speedometer.ino:55:
c:\users\rens\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\iom324p.h:751:0: note: this is the location of the previous definition
 #define RXCIE0  7
 
In file included from C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\cores\MCUdude_corefiles/Arduino.h:393:0,
                 from G:\projects\speedometer\odometer.h:1,
                 from G:\projects\speedometer\speedometer.ino:56:
C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\variants\standard/pins_arduino.h:522:0: warning: "UCPOL0" redefined
   #define UCPOL0  UCPOL
 
In file included from c:\users\rens\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\io.h:250:0,
                 from c:\users\rens\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\eeprom.h:38,
                 from C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\libraries\EEPROM\src/EEPROM.h:25,
                 from G:\projects\speedometer\speedometer.ino:55:
c:\users\rens\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\iom324p.h:754:0: note: this is the location of the previous definition
 #define UCPOL0  0
 
In file included from C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\cores\MCUdude_corefiles/Arduino.h:393:0,
                 from G:\projects\speedometer\odometer.h:1,
                 from G:\projects\speedometer\speedometer.ino:56:
C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\variants\standard/pins_arduino.h:523:0: warning: "UCSZ00" redefined
   #define UCSZ00  UCSZ0
 
[... 916 lines cut ...]
   ^~~~~~
[... cut ...]
Using library SPI at version 1.0 in folder: C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\libraries\SPI 
Using library SwitecX25-master in folder: D:\Users\Rens\Documents\Arduino\libraries\SwitecX25-master (legacy)
Using library Wire1 at version 1.0 in folder: C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\libraries\Wire1 
Using library Wire at version 1.1 in folder: C:\Users\Rens\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\3.0.1\libraries\Wire 
exit status 1

Compilation error: 'TIMSK3' was not declared in this scope
```

MCUdude, 2024-02-08:
> c:\users\rens\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\iom324p.h:748:0: note: this is the location of the previous definition

It looks to me like you're compiling the code for the ATmega324P instead of 324PB. Are you sure you have selected the 324PB under the variants menu?

rvoogd, 2024-02-09:
Thank you for your quick reply, it was indeed a stupid mistake that I somehow had ATmega324P selected instead of the ATmega324PB. No idea how that happened and why I hadn't realised.

Apologies for the mistake. Thanks for your help and for your work on this Arduino Core - it's very impressive and works well

MCUdude, 2024-02-09:
Great to hear you got it working!

> Thanks for your help and for your work on this Arduino Core - it's very impressive and works well

Thanks! I created most of these Arduino cores about 8 years ago, because proper 3rd party AVR cores didn't exist back then. My cores got way more traction than I intended, and I've just continued to maintain them over the years. They are pretty proven, and AFAIK used in many commercial applications as well. And with the new bootloader, [Urboot](https://github.com/stefanrueger/urboot), you get a feature-packed bootloader that's even smaller than Optiboot!

If you want to play with the bootloader, I'll recommend using [Avrdude 7.3](https://github.com/avrdudes/avrdude) directly, where you can play around and read/write any data or file to the flash or EEPROM.

```
$ avrdude -c urclock -p atmega324pb -P COM4 -t
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9517 (probably m324pb)

avrdude: processing -t interactive terminal
avrdude> read eeprom 0 256 # read 256 bytes from EEPROM starting from address 0
Reading | ################################################## | 100% 0.00 s 
0000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
avrdude> write eeprom 0 "Hello World EEPROM test" 0x11223344 'a' 'v' 'r'
Caching | ################################################## | 100% 0.00 s 
avrdude> read eeprom 0 256
Reading | ################################################## | 100% 0.00 s 
0000  48 65 6c 6c 6f 20 57 6f  72 6c 64 20 45 45 50 52  |Hello World EEPR|
0010  4f 4d 20 74 65 73 74 00  44 33 22 11 61 76 72 ff  |OM test.D3".avr.|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0050  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0070  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0080  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0090  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00b0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00c0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00d0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00e0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
avrdude> q
avrdude: synching cache to device ... 
Writing | ################################################## | 100% 0.00 s 

avrdude done.  Thank you.
```

closed 2024-02-09 by MCUdude