MCUdude/MightyCore #293: Atmega16A Serial can not use, until 1 sec delay.

happysoul, 2024-03-24:
arduino code
```
void setup() {
  Serial.begin(115200);
}

void loop() {
  if(Serial.available()){
    Serial.write(Serial.read());    
  }
}
```

python code
```
from time import sleep
from serial import Serial

serial = Serial(baudrate=115200, timeout=3, write_timeout=2)
# change your port here ##### 
serial.port = 'COM4'

try:
    serial.open()
except:
    print("exception")
    exit(0)

if serial.is_open:
    # need delay less then 1 second, wait for uart ready #### 
    sleep(1)
    serial.write('test\n\r'.encode())
    line = serial.readline()
    print(line)
else:
    print('Port not open')

```

If remove 'sleep(1)'  serial can not run correct.
Serial need 1 sec + ,then can used. I used 1 sed, 0.9 sec, 2 sec ....

soft: arduino 1.8.19  with last version MightyCore
hard: usb - ch340c - atmega16a

happysoul, 2024-03-24:
[image]
[image]

MCUdude, 2024-03-24:
Are you using a bootloader with auto reset circuity present (100n capacitor in series)?

happysoul, 2024-03-24:
[image]

MCUdude, 2024-03-24:
> Are you using a bootloader with auto reset circuity present (100n capacitor in series)?

Yes, you are. The bootloader has a 1-second timeout, and when you start your python application, the chips are reset by the DTR line and stay in bootloader mode for a second. So the behavior you're seeing is to be expected. Alternatively, you can remove the bootloader

happysoul, 2024-03-24:
Design in expected.  (lll￢ω￢)  got it ...
Thanks 
I initially thought this was a bug,  atmega8a did not need this 1 second

MCUdude, 2024-03-24:
No worries. Like I mentioned, you can remove the bootloader if that one second is crucial.

closed 2024-03-24 by MCUdude