MCUdude/MightyCore #294: Issue with uploading through Serial0 on custom PCB

The0Scorpion, 2024-03-25:
I have just received my new PCBs for my graduation project, but I'm having difficulty uploading via the USB to TTL, this the first time for me including a ch340 'USB to TTL chip and a USB port to program my PCB directly. 
After some fiddling, I got the PC to recognize the ch340C finally (it was a random soldering issue), and now when I type in Serial Moniter the Leds light up. 
Also when I upload a sketch that writes to serial the TX led lights up. 
The problem is I can't see the characters Sent from MCU in Serial, also The MCU can't see the data sent to it (I tested that with a simple if "Serial.available()>>> light a led" sketch)
So TLDR the MCU and CH340C are sending some data but they don't understand each other. 

the setup:
[image]

[image]

_*I'm using an Atmega32 instead of the 324 for development reasons 
*The caps C17,C13 are 150nf instead of 100nf because of local supply issues_ 

**what I tried:**

1. An external USB to but still didn't read any data, I think that is because if the ch340 being on the same data lines.
2. changing programming baud rate 
3. switching from the external crystal to internal crystal with multiple different frequencies.

**What am I doing wrong?**
should I continue diagnosing the PCB itself for soldering issues or is there is something fundamentally wrong with my design.

The0Scorpion, 2024-03-25:
Thanks for the hasty response, 
I realized my mistake before I saw your response and removed the LEDs completely and replaced the current limit with 0Ohm resistors and it worked!
I wanted to share the complete schematic, but my team won't allow that.
I did read your DIP-40 development board guide and other ch340c guides, but it was simply my mistake.
Also, I'm in the process of making both my software and Schematics more readable, Sorry for making you suffer through it. 
Everything works now! 
Other than a better-quality USB cable I'm ready to implement my software.
Here is pictures of the Actual PCB:
[image]
[image]
Thanks for help so much.

MCUdude, 2024-03-25:
Great to hear you figured it out. Good luck with your project!

closed 2024-03-25 by MCUdude