MCUdude/MightyCore #295: Just Got 2 From Norway, ... sorry but what is the use case of MightyCore other than ulisp

kwccoin, 2024-09-13:
I am collecting my ulisp badge BoM and one of them is the MightyCore.  See http://www.technoblogy.com/show?2AEE and 

```
Installing a bootloader

To program the Lisp Badge I now recommend using MCUdude's [MightyCore on GitHub](https://github.com/MCUdude/MightyCore).

```

I know it sound crazy but it would be years before I will do this (and it is many years when this project started).  The question hence is ... what is MightyCore and what is the use case of this?  

Sorry.  Crazy I know.  Buy first and ask what is that later is not the normal approach I do thing.  This is really the first.  Hence ...

What is MightCore?

MCUdude, 2024-09-13:
MightyCore is a 3rd party add-on to Arduino IDE, which lets you program the ATmega1284 and lots of other pin-compatible chips using Arduino IDE and the Arduino API.

You don't have to use it if you don't want to, but it's very convenient if you're already used to Arduinos in general.

kwccoin, 2024-09-14:
Many thanks for your prompt reply.  May I still waste of your time to ask even further.  Though my project would be years away, I am kind of mind person and hence to organize my thoughts and models.  I just want to fix in my mind where should I put this piece of hardware in my collection (of hardware/gear/mind space).  

I do STM32 and they either come with a programmer (in developer board) or I use my STlinkV2 or there is somewhere my J programmer.  I also have a programmer for some EExROM (say for my Ben Eater 6502 project).  

Arduino to me is for ESF32 and ATMxx processor and they use serial port to program ...

Sorry, I am totally confuse by this and what is the points of this.  Hope I am not sound silly or rude.  It is just totally a curve ball to me (even though I really do not understand base ball  :-)).

MCUdude, 2024-09-14:
Don't worry! You're not rude at all, but I'm struggling to figure out what the question is.

> Arduino to me is for ESF32 and ATMxx processor and they use serial port to program ...

The term "Arduino" is a bit ambiguous. It's a board, but it's also an ecosystem. The ESP32 wasn't originally hardware that were used in official Arduino products, but "Arduino support" were provided using Espressifs 3rd party addon.

The same goes for "plain" ATmegas as well. These are just microcontrollers, and nothing particularly fancy. You can upload code to them using an ISP programmer, or you can use a serial port if the microcontrollers already have a bootloader present. A bootloader is needed if you want to upload using a serial port. And the bootloader has to be flashed using an ISP programmer. The boards I sell on Tindie comes pre-programmed with a bootloader, so you don't need to use an ISP programmer with these if you don't want to.

MightyCore lets you use existing Arduino libraries and code for ATmega microcontrollers that has never been officially supported by Arduino IDE. And that's the thing. MightyCore, and my other Arduino cores lets you use practically _any_ ATmega with Arduino IDE, and they work just as well as a chip mounted to an Arduino UNO board.

If you want the following code snippet to "just work" on an ATmega1284P, you'll have to use MightyCore, becuase the avr-gcc compiler doesn't know what `pinMode` or `digitalWrite` is.

```c
// Arduino compatible code

#define LED_PIN 0

void setup() {
  pinMode(LED_PIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_PIN, HIGH);
  delay(1000);
  digitalWrite(LED_PIN, LOW);
  delay(1000);
}
```

If you want to do this using "native" libraries for AVR microcontrollers, the same code would look like this:

```c
// "Native" code
#define F_CPU 16000000L // Main clock speed
#include <avr/io.h>
#include <util/delay.h>

int main(void) {
  // Set PB0 as output
  DDRB |= (1 << PB0);

  while(1) {
    PORTB |= (1 << PB0); // Set PB0 high
    _delay_ms(1000); // Wait 1000 ms
    PORTB &= ~(1 << PB0); // Set PB0 low
    _delay_ms(1000); // Wait 1000 ms
  }  
  return 0; // Must never return
}
```

kwccoin, 2024-09-15:
Thanks.  Understand more.  I need to dig more into this loadbooter issue (at least for my STM32).  With that, may I close and wish you all the best with your business.

closed 2024-09-15 by kwccoin