MCUdude/MightyCore #296: Unable to burn boot loader ATMGA644PA on MacOS

sensorhost, 2024-12-14:
Hi, I have a board that I am unable to burn the boot loader to.
I have been writing the bootloader to other boards today (using Minicore and 328P) successfully so I don't think it is my programmer (AVRISP Mk II clone - the one in a semi transparent blue case) and have been using it for several years without fault.

Of note is that I have yesterday updated my MacBook Pro (from an Intel version to an M4 Pro), but have been using it successfully with MiniCore and the 328P earlier today.

Error from Arduino IDE below for reference.

Not sure why, but it shows MegaCoreX on the first line for avrdude?? plus several other warnings and errors.

Any help is appreciated.

```
"/Users/gjt211/Library/Arduino15/packages/MegaCoreX/tools/avrdude/8.0-arduino.1/bin/avrdude" "-C/Users/gjt211/Library/Arduino15/packages/MightyCore/hardware/avr/2.1.3/avrdude.conf" -v -patmega644p -cstk500 -P /dev/cu.wchusbserial2141120 -e -Ulock:w:0x3f:m -Uefuse:w:0xfd:m -Uhfuse:w:0b11010111:m -Ulfuse:w:0xf7:m
Avrdude version 8.0-arduino.1
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /Users/gjt211/Library/Arduino15/packages/MightyCore/hardware/avr/2.1.3/avrdude.conf
Warning: programmer wiring fails to specify prog_modes = PM_...; [/Users/gjt211/Library/Arduino15/packages/MightyCore/hardware/avr/2.1.3/avrdude.conf:351]
Warning: programmer arduino fails to specify prog_modes = PM_...; [/Users/gjt211/Library/Arduino15/packages/MightyCore/hardware/avr/2.1.3/avrdude.conf:358]
Error: unknown token [/Users/gjt211/Library/Arduino15/packages/MightyCore/hardware/avr/2.1.3/avrdude.conf:393]
Warning: programmer avrftdi fails to specify prog_modes = PM_...; [/Users/gjt211/Library/Arduino15/packages/MightyCore/hardware/avr/2.1.3/avrdude.conf:393]
Error: syntax error [/Users/gjt211/Library/Arduino15/packages/MightyCore/hardware/avr/2.1.3/avrdude.conf:393]
Error: unable to process system wide configuration file /Users/gjt211/Library/Arduino15/packages/MightyCore/hardware/avr/2.1.3/avrdude.conf
Failed chip erase: uploading error: exit status 1

```

[image]

MCUdude, 2024-12-14:
Try uninstalling MightyCore and installing it again. It looks like it's using Avrdude 8.0 with an older configuration file.

sensorhost, 2024-12-14:
A bit of an update, but still not working.

Identified that Mightycore was manually installed, not via the boards manager.
Added to the boards manager, then done an update to the latest version.
Tried to burn the boot loader again and the following is the output.

Just to double check my sanity, I just tried again with Minicore to reprogram one on the M328 boards I done earlier today, now for some reason I am unable to burn the boot loader to my M328 boards also (actually programmed about 50 of them this morning successfully) so now I am lost.

```
"/Users/gjt211/Library/Arduino15/packages/MightyCore/tools/avrdude/8.0-arduino.1/bin/avrdude" "-C/Users/gjt211/Library/Arduino15/packages/MightyCore/tools/avrdude/8.0-arduino.1/etc/avrdude.conf" -v -patmega644p -cavrispmkii  -e -Ulock:w:0xff:m -Uefuse:w:0xfd:m -Uhfuse:w:0xd7:m -Ulfuse:w:0xf7:m
Avrdude version 8.0-arduino.1
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /Users/gjt211/Library/Arduino15/packages/MightyCore/tools/avrdude/8.0-arduino.1/etc/avrdude.conf
User configuration file /Users/gjt211/.avrduderc does not exist

Using port            : usb
Using programmer      : avrispmkII
Error: did not find any USB device usb (03eb:2104)
Error: unable to open port usb for programmer avrispmkII

Avrdude done.  Thank you.
Failed chip erase: uploading error: exit status 1
```

sensorhost, 2024-12-14:
Just tried uninstalling MightyCore, then quit and restarted Arduino IDE 2.3.4.
Then re-installed using boards manager, still same problem sorry.

```
"/Users/gjt211/Library/Arduino15/packages/MightyCore/tools/avrdude/8.0-arduino.1/bin/avrdude" "-C/Users/gjt211/Library/Arduino15/packages/MightyCore/tools/avrdude/8.0-arduino.1/etc/avrdude.conf" -v -patmega644p -cavrispmkii  -e -Ulock:w:0xff:m -Uefuse:w:0xfd:m -Uhfuse:w:0xd7:m -Ulfuse:w:0xf7:m
Avrdude version 8.0-arduino.1
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /Users/gjt211/Library/Arduino15/packages/MightyCore/tools/avrdude/8.0-arduino.1/etc/avrdude.conf
User configuration file /Users/gjt211/.avrduderc does not exist

Using port            : usb
Using programmer      : avrispmkII
Error: did not find any USB device usb (03eb:2104)
Error: unable to open port usb for programmer avrispmkII

Avrdude done.  Thank you.
Failed chip erase: uploading error: exit status 1
```

sensorhost, 2024-12-14:
Looks like I found the problem, as usual it is my fault.
For some reason, the AVRISP was not showing up in the System Information, my industrial grade USB hub that has worked flawlessly for the last 18 months decided even though my Mac could see it, it didn't want to allow the AVRISP to work. A power cycle on the HUB fixed the problem.

closed 2024-12-14 by sensorhost