MCUdude/MightyCore #297: Atmega32A pins 18-21, will not funcion with Mightycore

BNSF-FAN, 2025-03-27:
I am using simple on/off commands, and it doesn't even respond. I am relatively new to Arduino series, so, if someone could help me with this, it would be awesome, and very much appreciated....

Best regards;;

Taz...

MCUdude, 2025-03-27:
Can you share a bare minimum version om your code, just for me tro try to reproduce the issue?

BNSF-FAN, 2025-03-27:
https://www.arduino.cc/en/Tutorial/BuiltInExamples/Blink
*/

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(21, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(21, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(1000);                      // wait for a second
  digitalWrite(21, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);                      // wait for a second
}

BNSF-FAN, 2025-03-27:
I am told that the functions assigned to these 4 pins, cannot be used as a digital output, until these 4 functions are disabled.
Do i need to set flags or bits on control registers to disable these? How would I go about disable these?

BNSF-FAN, 2025-03-27:
I need to disable the JTAG functions, in order to use these 4 pins as I/O. How do I accomplish this?
Many thanks!

Taz

BNSF-FAN, 2025-03-27:
hERE IS THE FIX.....

https://forum.arduino.cc/t/jtag-disable-problem/289448/5

MCUdude, 2025-03-27:
Just burn the bootloader using Arduino IDE. This will automatically clear the JTAG fuse

closed 2025-03-27 by MCUdude