MCUdude/MightyCore #299: Programmer and Bootloader menu option no longer available

paulfer747, 2025-04-15:
Programmer and Bootloader menu option no longer available.

Mightycore ver: 3.0.3

Arduino IDE Version: 2.3.5
Date: 2025-04-02T13:16:49.885Z
CLI Version: 1.2.0

MCUdude, 2025-04-15:
Sorry, but I'm not able to reproduce the issue. I have IDE 2.3.6 installed, and MightyCore v3.0.3.

[image]

paulfer747, 2025-04-15:
This seems to be a IDE issue. I re-installed everything, but still the same. 

For info: I can create shortcuts to both the programmer and the burn bootloader by using FILE>ADVANCED>KEYBOARD SHORTCUTS, and they work, they just dont appear in the menu bar.

As an aside, I Also, get this message when updating your Minicore to 3.1.0  :
Failed to install platform: 'MiniCore:avr:3.1.0'.
Error: 13 INTERNAL: Cannot install tool MiniCore:dw-tools@2.1.6: testing local archive integrity: testing archive checksum: archive hash differs from hash in index

MCUdude, 2025-04-15:
> For info: I can create shortcuts to both the programmer and the burn bootloader by using FILE>ADVANCED>KEYBOARD SHORTCUTS, and they work, they just dont appear in the menu bar.

You should probably report this to the Arduino developers (or perhaps @per1234 has a clue). I've not done any major, structural changes to the tools menu for a long time.

> As an aside, I Also, get this message when updating your Minicore to 3.1.0 :

The MiniCore issue is a known issue. I've rolled back v3.1.0, and a new v3.1.0 release will be available soon. Meanwhile, please use MiniCore v3.0.4 instead.

The reason why is that I'm working on adding full debug support for Arduino IDE 2, using a Microchip EDBG based debugger such as the Atmel ICE.

closed 2025-04-15 by paulfer747