MCUdude/MightyCore #300: urbootPageWrite is not working on 1284p

NilocEidlog, 2025-05-13:
I cannot get urbootPageWrite working on a 1284p.

ispgmwritepage076_ur() is returning false no mater what i do.

Bootloader is the standard urboot_{build.mcu}_pr_ee_ce.hex

lock_bits = 0x3F
hfuse = 0xDE
lfuse = 0x7F
efuse = 0xFF

This should allow SPM in bootloader or app , but calling page write on an unused area is crashing and ispgmwritepage076_ur() is always returning false.

NilocEidlog, 2025-05-13:
Mighty Core 3.0.3 and Arduino 1.8.18

stefanrueger, 2025-05-13:
See my take [here](https://github.com/stefanrueger/urboot.hex/issues/6)

stefanrueger, 2025-05-14:
And if setting the correct fuses does not work either, try downloading and burning the bootloader directly from [urboot.hex](https://github.com/stefanrueger/urboot.hex/tree/main/mcus/atmega2560/watchdog_1_s/autobaud/uart0_rxe0_txe1/no-led).